peted-davis/WeatherFlow_PiConsole #156: console is showing NC Rain Nearcast value and not actually

blainejt, 2025-07-12:
the console is not displaying the real rain value but is showing the  NC Rain Nearcast value, you can see in the attached images (3 one of console, one of app with NC rain turned on , one of app with NC Rain turned off) that
the console value matches the apps rain value when  NC Rain Nearcast is turned on and not the value when turned off.

i am running wfpi console v 25.2.1

is there a setting not in the UI to make it show NC rain and not real rain?

[image]

[image]

[image]

peted-davis, 2025-07-13:
The data returned in each Tempest websocket packet was recently changed to contain NC rainfall values rather than raw rainfall values. There is no option to change this in wfpiconsole. Does toggling NC rain off in the app affect what is shown on the console? You might need to reboot the console after toggling NC rain off

blainejt, 2025-07-13:
i never have had  NC rain on , i only turned it on to see if that was the issue.

peted-davis, 2025-07-13:
Ok, understood. Unfortunately there is nothing I can do unless WeatherFlow make a change to the API. Alternatively, you can run the console in UDP mode

closed 2025-07-13 by peted-davis