peted-davis/WeatherFlow_PiConsole #160: Updates for Debian Trixie

reubenfarrelly, 2025-09-01:
Firstly, I will state that I am well are that Debian Trixie has not been officially released for the Raspberry Pi yet, and that I accept that what I have attempted to do is not officially supported.  But I wanted to share my experiences so far, as it may be possible to implement some of these updates now which apply even for earlier versions of Raspberry Pi OS which will make the update in the future more seamless and/or existing updates more resilient.

After update to Debian Trixie on the RPI5, wfpiconsole would not start, and required some manual steps to get it to build and run.  The steps I did were:

Edit wfpiconsole.sh:

- PYTHON_MODULES=(websockets==11.0.3
                numpy==1.26.0    <<< === Update this to numpy 2.2.4 as this is what ships with Trixie.  Could this be changed now, to require at a minimum version 1.26 but allow any higher version?

- KIVY_VERSION="2.2.0"    <<< ===  changed to 2.3.1 which is the latest version on the kivy website - system failed to build/update with 2.2.0 but worked fine with 2.3.1.  Could this be updated now, and still be valid, even for earlier versions of Raspberry Pi OS?

Remove .kivy directory in /home/wfpiconsole
Remove .local directory in /home/wfpiconsole   (unclear if this was really required)
Remove venv directory in /home/wfpiconsole/wfpiconsole

After completing those steps, the console was then able to start successfully and appears to run normally.  With the basic testing I have done so far it appears that this is all that was required post update.

peted-davis, 2025-09-01:
This looks possible and quick testing suggests it shouldn't break the install on earlier versions. I will try and get an update out soon

reubenfarrelly, 2025-09-02:
So a bit more of an update on this as I've been tinkering today.  I've updated even more of the python modules - all to the latest version, and still everything seems to be working completely normally, even after a complete new install:

# Python modules and versions
PYTHON_MODULES=(websockets==15.0.1
                numpy==2.3.2
                pytz==2025.2
                tzlocal==5.3.1
                ephem==4.2
                packaging==25.0
                cryptography==45.0.7
                pyOpenSSL==25.1.0
                certifi==2025.8.3)

Some of these updates (numpy for sure) is required for python 3.13 support which is in Trixie.  Earliver versions did not support python 3.13.

Other changes I had to make:

* Remove libatlas-base-dev - this dependency/package is no longer in Trixie, seems Debian have removed this package.  See:
https://alioth-lists.debian.net/pipermail/debian-science-maintainers/2023-November/114473.html for an explanation.

* Update KIVY_VERSION="2.2.0"  to 2.3.2    

Now what would be ideal, is if all these package version requirements could be updated in the install script, and then that this has no detrimental impact on anyone still on Bookworm or earlier releases.  I'd like to think that no one and no package has broken backward compatibility with these python updates - if this is the case then this will mean it's relatively easy to support Trixie.

Log from startup looks relatively clean too:

[INFO   ] [Logger      ] Record log in /home/wfpiconsole/.kivy/logs/kivy_25-09-02_3.txt
[INFO   ] [Kivy        ] v2.3.1
[INFO   ] [Kivy        ] Installed at "/home/wfpiconsole/wfpiconsole/venv/lib/python3.13/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.13.5 (main, Jun 25 2025, 18:55:22) [GCC 14.2.0]
[INFO   ] [Python      ] Interpreter at "/home/wfpiconsole/wfpiconsole/venv/bin/python3"
[INFO   ] [Logger      ] Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
[INFO   ] [Config      ] Verifying station details
[INFO   ] [Factory     ] 195 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Window      ] Provider: sdl2
[INFO   ] [GL          ] Using the "OpenGL ES 2" graphics system
[INFO   ] [GL          ] Backend used <sdl2>
[INFO   ] [GL          ] OpenGL version <b'3.1 Mesa 25.0.7-2+rpt1'>
[INFO   ] [GL          ] OpenGL vendor <b'Broadcom'>
[INFO   ] [GL          ] OpenGL renderer <b'V3D 7.1.7.0'>
[INFO   ] [GL          ] OpenGL parsed version: 3, 1
[INFO   ] [GL          ] Shading version <b'1.40'>
[INFO   ] [GL          ] Texture max size <4096>
[INFO   ] [GL          ] Texture max units <24>
[INFO   ] [Window      ] auto add sdl2 input provider
[INFO   ] [Window      ] virtual keyboard not allowed, single mode, not docked
[INFO   ] [Text        ] Provider: sdl2
[INFO   ] [GL          ] NPOT texture support is available
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
WARNING: running xinput against an Xwayland server. See the xinput man page for details.
[INFO   ] [ProbeSysfs  ] device match: /dev/input/event5
[INFO   ] [HIDInput    ] Read event from </dev/input/event5>
[INFO   ] [Base        ] Start application main loop
[INFO   ] [HIDMotionEvent] using <11-005d Goodix Capacitive TouchScreen^@>
[INFO   ] [HIDMotionEvent] <11-005d Goodix Capacitive TouchScreen^@> range ABS X position is 0 - 719
[INFO   ] [HIDMotionEvent] <11-005d Goodix Capacitive TouchScreen^@> range ABS Y position is 0 - 1279
[INFO   ] [HIDMotionEvent] <11-005d Goodix Capacitive TouchScreen^@> range position X is 0 - 719
[INFO   ] [HIDMotionEvent] <11-005d Goodix Capacitive TouchScreen^@> range position Y is 0 - 1279
[INFO   ] [Websocket   ] 2025-09-02 20:26:50 - Opening connection
[INFO   ] [Websocket   ] 2025-09-02 20:26:51 - Connection open

peted-davis, 2025-09-06:
OK, great. Same again: please delete the existing virtual environment within the wfpiconsole directory and then fetch a copy of the new install script using wget https://raw.githubusercontent.com/peted-davis/WeatherFlow_PiConsole/refs/heads/develop/wfpiconsole.sh.

Make the new .sh file executable and run ./wfpiconsole.sh install. Hopefully this should complete without any errors or warnings

peted-davis, 2025-09-21:
This has now been rolled into v25.9.1

closed 2025-09-21 by peted-davis